robotology/mesh-support #1722: Neck cable broken

SimoneMic, 2024-01-25:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The neck tendon broke during a sudden movement of the torso due to an unexpected singularity during test grasping.

### Detailed context

[image]

GiacomoVassallo13, 2024-01-25:
Hi the broken tendon has been replaced.  i ttend a fedback of the testin in softwere.

cc: @Fabrizio69 @sgiraz @AntonioConsilvio @fbiggi

mebbaid, 2024-01-25:
We didn't have any issues from the neck during our experiment that included walking with the neck joints controlled.

sgiraz, 2024-01-25:
Ok, thanks @mebbaid. Closing!

closed 2024-01-25 by sgiraz