robotology/mesh-support #1723: Right thumb distal tendon broken

fedeceola, 2024-01-25:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

After fixing the problem in #1702, the distal tendon of the right thumb broke. @GiacomoVassallo13 fixed it but robot cannot perform the calibration of the right hand fingers. 

### Detailed context

After fixing the problem in #1702, the distal tendon of the right thumb broke. @GiacomoVassallo13 fixed it but robot cannot perform the calibration of the right hand fingers.

GiacomoVassallo13, 2024-01-26:
Hi. I correct the title of the issue because the broken tendon was the distal not the proximal. The distal joint of the thumb has been replaced and the joint was been tested. I report a broken electric cable on my little finger and problems in calibrating the joints @AntonioConsilvio. 

cc; @Fabrizio69

AntonioConsilvio, 2024-03-06:
All joints are working now! See this [comment](https://github.com/robotology/icub-tech-support/issues/1702#issuecomment-1980848356).