robotology/mesh-support #1725: Broken cable in robot neck

G-Cervettini, 2024-01-25:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Robot fell while walking with the crane and the head touched one of the safety slings, breaking one of the motor cables.

### Detailed context

Robot fell while walking with the crane and the head touched one of the safety slings, breaking one of the motor cables.

G-Cervettini, 2024-01-25:
cc @GiulioRomualdi

sgiraz, 2024-01-25:
Hi @G-Cervettini,

Do you have any additional detailed context to share with us that may explain the reason of the fault?
Any log? Any videos?

We recognize the need to enhance the neck assembly/system, and these information could prove invaluable in achieving that goal.

GiulioRomualdi, 2024-01-26:
The robot fell and the crane belt blocked the head

GiulioRomualdi, 2024-01-26:
If you want I can show you better what happened in the robot arena

GiacomoVassallo13, 2024-01-26:
Hi. The broken neck cable has been replaced. We attend the testing feedback,.

cc: @Fabrizio69 @sgiraz @fbiggi

G-Cervettini, 2024-01-26:
> Hi @G-Cervettini,
> 
> Do you have any additional detailed context to share with us that may explain the reason of the fault? Any logs? Any videos?
> 
> We recognize the need to enhance the neck assembly/system, and these information could prove invaluable in achieving that goal.

[neckCableFailure_ergoCub.zip](https://github.com/robotology/icub-tech-support/files/14067058/neckCableFailure_ergoCub.zip)

There is also a video on `ergocub001-lap`, you can find it as `robot_logger_device_2024_01_25_18_10_07_jabra_rgb.mp4`

sgiraz, 2024-01-26:
Hi @GiulioRomualdi, @G-Cervettini,

Unfortunately, I'm working remotely until next Tuesday and I cannot access to the robot arena.
I tried to open the .zip file posted above, but I cannot see useful information. There are only 2 file ( `robot_logger_device_2024_01_25_18_10_07.mat` and  `robot_logger_device_2024_01_25_18_10_07.md` ) 
Could you post the video and the log of the `yarprobotinterface` here, please?

>[!Warning]
>I strongly advise against exchanging `.zip` files in the issues threads as it is an unreliable method for sharing materials

sgiraz, 2024-02-15:
We are aware of the critical issues with the neck tendons and we are working to improve it as much as possible.

For now I proceed to close this issue, since the tendons have been replaced successfully.

closed 2024-02-15 by sgiraz