robotology/mesh-support #1726: r_knee, l_hip_yaw and l_knee went in hardware fault while walking

GiulioRomualdi, 2024-01-29:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Given the upcoming demo we tested the usual walking-controller. While performing the steps some joints on the robot leg went in Hardware Fault.

### Detailed context

The robot was walking using the stable controller we use for the demo and it was performing the classical (turning in place and walking forward) steps.
At a certain point the robot felt down because one of the joint of the robot went in HF. We were not able to detect which one. We opened the motorgui and we notice that r_knee, l_hip_yaw and l_knee were in HF

### Additional context

We have the txt of the yarprobotinterface 
[log_ergocub-torso_yarprobotinterface_2721_knee_falling.txt](https://github.com/robotology/icub-tech-support/files/14082492/log_ergocub-torso_yarprobotinterface_2721_knee_falling.txt)

while these are the screen of the motorgui

### left leg
They seem to be not informative
[image]

### right leg

An overcurrent was detected 

[image]

### How does it affect you?

We have two demos in this week:
- https://github.com/ami-iit/lab-events-demos/issues/297
- https://github.com/ami-iit/lab-events-demos/issues/296

GiulioRomualdi, 2024-01-29:
cc @mebbaid @S-Dafarra

GiulioRomualdi, 2024-01-29:
We are going to reboot the robot (boards and pc) and try again

GiulioRomualdi, 2024-01-29:
We rebooted the robot we ran the interface and the interface crashed with following message 
```
[ERROR] yarprobotinterface intercepted a segmentation fault caused by a faulty plugin:
[ERROR] /usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_ZN4yarp2os4impl10BottleImpl7subCodeEv+0x54) [0x7f9078c43ed4]
Trace requested at /usr/local/src/robot/robotology-superbuild/src/YARP/src/yarprobotinterface/Module.cpp:77 by code called from:
/usr/local/src/robot/robotology-superbuild/build/install/lib/libYARP_os.so.3(_Z16yarp_print_traceP8_IO_FILEPKcj+0x3b) [0x7f9078bb90cb]
```
[log_ergocub-torso_yarprobotinterface_3166_segs_fault.txt](https://github.com/robotology/icub-tech-support/files/14082644/log_ergocub-torso_yarprobotinterface_3166_segs_fault.txt)

I'm going to open an issue for this still I don't know if could be related

AntonioConsilvio, 2024-05-07:
Hi! Agreed with @GiulioRomualdi, this issue will be closed without a real intervention for two reasons:

- The first is that despite the information provided, the `generic motor error` does not allow us to have an idea of what could be causing the problem.

- The second is that the problem has not recurred.

In addition, the diagnostics have improved over this time, so if the problem were to recur, it would show more accurate errors and not `generic motor error`.

Closing!

closed 2024-05-07 by AntonioConsilvio