robotology/mesh-support #1727: Unable to read the data coming from the BMS/BAT

gabrielenava, 2024-01-29:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

See a similar issue on Ergocub https://github.com/robotology/icub-tech-support/issues/1715

### Detailed context

Following [this comment](https://github.com/robotology/icub-tech-support/issues/1715#issuecomment-1901937104) in a similar issue for Ergocub (https://github.com/robotology/icub-tech-support/issues/1715): 

>It seems to me as well, please @gabrielenava open an issue for fixing this, thanks.

the same problem is occurring on iRonCub3, so if the problem is fixed for Ergocub we might replicate the solution to iRonCub

cc @S-Dafarra @DanielePucci

sgiraz, 2024-01-29:
cc @MSECode

MSECode, 2024-01-29:
Oks, need to check the cabling of the CAN connection between EMS02 and BMS (if BMS is present, I think so) and EMS01 and BAT and then check again. Some data on the BAT are shown always as none, such as the current, since we are not currently sending it since it is not accurate. There'll be soon an update where we are going to send it anyways.

gabrielenava, 2024-05-02:
Thanks @AntonioConsilvio! we'll test it in the next experimental campaign!

gabrielenava, 2024-06-18:
cc @davidegorbani

gabrielenava, 2024-06-18:
sorry for the delay @AntonioConsilvio, we finally tested what you suggested in https://github.com/robotology/icub-tech-support/issues/1727#issuecomment-2075204918

we got this output from the port:

[image]

if you think it is ok despite the NaN, we can close the issue!

MSECode, 2024-06-19:
@gabrielenava that is correct. Since you are reading data from the BAT, 4th value is always NaN that is the battery pack temperature and it is not available for BAT. And the 3rd value is available only if the BMS is connected because is the battery pack state of charge.
So that is fine and I think you can close the issue.

Check the details here: https://icub-tech-iit.github.io/documentation/battery_boards/bat_board/bat_board/#data-displayed-on-the-yarp-port

AntonioConsilvio, 2024-06-19:
Good job @gabrielenava and thanks for the explanation @MSECode, as everything is as expected I will continue by closing the issue!

I would like to remind that the battery (and therefore the BMS) is currently not present on iRonCub3, so the BAT behaviour makes perfect sense. Closing!

closed 2024-06-19 by AntonioConsilvio