robotology/mesh-support #1729: Face expressions doesn't work

martinaxgloria, 2024-01-30:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

Face expressions stopped to work.

### Detailed context

At the startup, the LED of the face expressions don't turn on anymore. In the past days, the cover of the head was opened, so maybe a cable was disconnected.

[image]

AntonioAzocar, 2024-01-31:
Hello! I redid the cable (SPI1_3) connecting the **RFE** to the **Left Eyebrow** board and the face expressions are working again!
Thanks @sgiraz and @AntonioConsilvio 

[image]

sgiraz, 2024-02-01:
Well done @AntonioAzocar! 🚀

Closing as solved.

closed 2024-02-01 by sgiraz