robotology/mesh-support #1730: Neck broken

DanielePucci, 2024-01-30:
### Robot Name 🤖

None

### Request/Failure description

Kindly ask to repair the neck cable. This is very important in view of tomorrow's demos 
* https://github.com/ami-iit/lab-events-demos/issues/296
* https://github.com/ami-iit/lab-events-demos/issues/297

### Detailed context

We were running some tests on the robot and the torque control seemed to not ask for enough torque around the torso pitch. So, the torso went forward and the robot had a flick (colpo di frusta) breaking the neck.

[image]

sgiraz, 2024-01-31:
This morning @AntonioAzocar @fbiggi repaired the tendon.

DanielePucci, 2024-02-01:
Thanks a lot @sgiraz. I think we can close this issue then.

closed 2024-02-01 by DanielePucci