robotology/mesh-support #1731: MTB4c Fingetip Firmware

randaz81, 2024-01-30:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

- Check the current status of the board. Is it still able to read the fingertips?
- Investigate why the Firmwareupdater is not able to program the mtb4 board of the left hand fingertips with the standard icub-firmware-build firmware (EMS Board 10.0.1.3)
- We would like to obtain a clear procedure to easily program the board (we have to switch multiple times between the standard firmware and the devel firmware)

### Detailed context

We did a lot of unsuccessful tests trying to write the Xela firmware on the MTB4 board.

Since this operation failed, we tried to understand if the problem was that specific firmware version, and we tried to program the mtb4 with the standard mtb4c firmware downloaded from icub-firmware-build 
(https://github.com/robotology/icub-firmware-build/tree/master/CAN/mtb4c). Also in this case the operation failed.

We tried to do this operation:
- passing through the EMS (via ethernet)
 -connecting directly to the CAN. 
 In both cases **the FirmwareUpdater suddenly terminates the operation without actually writing the firmware.**
 
The board now is still visible from the FirmwareUpdater but I am not sure if it is still working or not. 

I was also not able to put it in bootloader mode (resetting the EMS) to recover its status to the standard firmware version.

sgiraz, 2024-03-15:
Hi @randaz81,

In the last days @Gandoo installed the new fingertips and replaced the `mtb4-eb2`:

[image]

> * We would like to obtain a clear procedure to easily program the board (we have to switch multiple times between the standard firmware and the devel firmware)

The procedure is described in https://icub-tech-iit.github.io/documentation/icub_firmware/firmware/firmware.

In a nuts, the user have to:
1. put in ETH Maintenance mode the ETH board involved in the CAN network where the CAN board is connected too
2. launch a `discover` w/ the ETH board selected
3. upload the new Firmware  with the target CAN board selected 
4. power off and power on the robot.

> Since this operation failed, we tried to understand if the problem was that specific firmware version, and we tried to program the mtb4 with the standard mtb4c firmware downloaded from icub-firmware-build

⚠️ [`mtb4`][1] is different from [`mtb4c`][2], be sure to flash the right FW Application.

Since your feedback are positive, I would consider the issue solved, right?

cc @AntonioConsilvio @Fabrizio69 

[1]: https://github.com/robotology/icub-firmware-build/tree/master/CAN/mtb4
[2]: https://github.com/robotology/icub-firmware-build/tree/master/CAN/mtb4c

randaz81, 2024-03-16:
We can close here, since the relevant discussion was moved to:
https://github.com/icub-tech-iit/tickets/issues/3344
https://github.com/icub-tech-iit/tickets/issues/3388

closed 2024-03-16 by randaz81