robotology/mesh-support #1732: Left leg joint jumped loosing the calibration

isorrentino, 2024-01-30:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While having the robot on the ground one of the joints of the left leg had a jump, moving to a new position. After this jump, the same encoder values for the two legs, result in a different physical configuration.

### Detailed context

Photos here:

[image]
[image]
[image]
[image]

cc @DanielePucci @maggia80 @GiulioRomualdi

DanielePucci, 2024-01-30:
We hoped that the jump was due to the AMO, but actually @isorrentino checked and we already using the LCORE. So, it seems that the problems are due to the LCORE measurements. Then, with @GiulioRomualdi, we re-enabled the AMO and we tested the walking **that started vibrating much more**. 

Clearly, if there is a problem in the LCORE, the field oriented control is going to be problematic.

@isorrentino suggested that the LCORE might be dirty: could this be the case @sgiraz @AntonioConsilvio @Fabrizio69 ?

sgiraz, 2024-01-31:
> Clearly, if there is a problem in the LCORE, the field oriented control is going to be problematic.

cc @ale-git @valegagge

sgiraz, 2024-01-31:
@ale-git, @AntonioConsilvio and I discussed F2F about the LCORE problem.

It is possible to decrease the tolerance in https://github.com/robotology/robots-configuration/blob/08f2a6ecc3c9d4cc2dad6b4524775b9f8f97e9a2/ergoCubSN001/hardware/motorControl/right_leg-eb6-j0_3-mc_service.xml#L51 to `2.1` (the minimum value) to see if a "dirty encoder" warning is detected.

We'll give it a try asap.

AntonioConsilvio, 2024-02-05:
Hi! The AMO cable has been fixed in this [issue](https://github.com/robotology/icub-tech-support/issues/1742):

It could be that the problem described in this issue was caused by abnormal behaviour of the AMO.
So, I would ask for feedback if the problem has happened again.

Thank you!

cc @isorrentino @GiulioRomualdi

AntonioConsilvio, 2024-02-14:
Since the robot completed the Farnesina's Demo successfully, without showing this problem anymore, I think that we can close this issue!

cc @isorrentino

closed 2024-02-14 by AntonioConsilvio