robotology/mesh-support #1736: Broken tendon of the neck

LoreMoretti, 2024-01-31:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

During some tests on torque control with the robot on the pole, the torso suddenly felt forward and caused the tendon of the neck to break.

cc @DanielePucci 

### Detailed context

We are still investigating why the torso felt forward, since it was not expected.

icub-tech already repaired the robot for a demo.

sgiraz, 2024-01-31:
Hi @LoreMoretti,

@AntonioConsilvio and @fbiggi just repaired the tendon.
In this case, the tendon's fault was due to the collision with the user in front of the robot after an unexpected fall forward.

>[!Warning]
>Using the torque control on the robot could be very dangerous. Be careful keeping yourself away enough to avoid unwanted collisions.

cc @DanielePucci @Fabrizio69

DanielePucci, 2024-01-31:
Thanks a lot @sgiraz @fils99 @AntonioConsilvio. We are trying to understand the root cause of this fast motion

closed 2024-02-01 by DanielePucci