robotology/mesh-support #1738: Robot going in "fake" overtemperature

DanielePucci, 2024-02-01:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The `left_hip_roll` goes to overheating, which seem a false positive. 

### Detailed context

After deploying the firmare for checking the temperature on the left leg in https://github.com/MSECode/icub-firmware/tree/fix/temperatureReading, @Miche19 and @mebbaid were trying to walk. After few steps, the robot fell 

https://github.com/robotology/icub-tech-support/assets/5045846/2801b563-4a1e-4ac4-9dd7-390d7fadd5f9

By checking at the yarpmotorgui, it seems an overheating
[image]

However, it seems a `false positive` since by putting the joint in idle and back to run it immediately gets back to run.

We don't have additional temperature data.

maggia80, 2024-02-01:
I would check the value of the temperature because it goes down quite rapidly.

https://icub-tech-iit.github.io/documentation/temperature_sensors/general/overview/?h=temper#how-to-read-the-motor-temperature

DanielePucci, 2024-02-01:
This issue seems that was related to https://github.com/robotology/icub-tech-support/issues/1739. After cleaning the LCORE, we no longer had the overheating problem. The robot walked for about five minutes and the temperatures look consistent.
[image]

So, actually the over temperature we got in the video above might have been true, and the protection system most likely worked!

On a side note, after https://github.com/robotology/icub-tech-support/issues/1739, the robot started walking much better and with less oscillations.

So, closing.

@GiulioRomualdi @isorrentino

closed 2024-02-01 by DanielePucci