robotology/mesh-support #1740: Right thumb got detached from the hand

GiulioRomualdi, 2024-02-01:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While testing the torque balancing, the robot collapsed; we pushed the fault button. During the process of catching it, the right thumb became detached from the hand.

### Detailed context

Probably there are some cables broken and the grub skewing got damaged.

fbiggi, 2024-02-02:
Inserted LOCTITE 648 glue on S3-8--_-_I2338_B.
I await info after movement tests

[image]
[image]

@sgiraz @Fabrizio69 @maggia80

GiulioRomualdi, 2024-02-02:
Thank you, @fbiggi!

fbiggi, 2024-02-02:
@GiulioRomualdi 
to position your thumb as you want for your experiments you must loosen the 2 screws as in the photo
The plug that connects the thumb to the palm must remain stationary
Please pass the info to your team

[image]

@sgiraz @Fabrizio69 @maggia80 @AntonioConsilvio

sgiraz, 2024-02-02:
Thanks @fbiggi for repairing the thumb! 🚀

I acknowledge that the broken thumb was an accident.

Anyway, consider the [suggestion above](https://github.com/robotology/icub-tech-support/issues/1740#issuecomment-1923247973) posted by @fbiggi in case you need to reposition the thumb-axis-rotation (not actuated).

Closing!

cc @GiulioRomualdi @AntonioConsilvio @SimoneMic @PasMarra @randaz81 @DanielePucci @Nicogene @Fabrizio69

closed 2024-02-02 by sgiraz