robotology/mesh-support #1742: right hip pitch not calibrating

LoreMoretti, 2024-02-02:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The right hip pitch does not calibrate.

It follows the yarp logger file.

### Detailed context

I tried to restart the motors and repeat the calibration.

I also tried to calibrate the individual joint via `yarpmotorgui --calib`.

But still it is not working:
[image]

### Additional context

[right_leg_eb6_not_configured.log](https://github.com/robotology/icub-tech-support/files/14139207/right_leg_eb6_not_configured.log)

[image]

S-Dafarra, 2024-02-05:
cc @maggia80 @DanielePucci

AntonioConsilvio, 2024-02-05:
Hi! The problem was due to a damaged AMO cable (RLS0).  
@Gandoo fixed it and I tested the robot.

It works properly now! Closing!