robotology/mesh-support #1745: Right jetpack FT sensor not detected

gabrielenava, 2024-02-05:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The right jetpack FT sensor is not detected at startup. The startup phase fails with this error:

```

	4,917304	ERROR	yarp.dev.PolyDriver	right_arm_jetpack-eb3-j0_1-strain	Driver <embObjMultipleFTsensors> was found but could not open
	4,917322	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=55s 433m 344u : CFG: theFTservice failed CAN discovery
	4,917329	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=55s 433m 453u : CFG: theFTservice failed CAN discovery
	4,917340	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=55s 433m 667u : CFG: CANdiscovery cannot find  1 missing eobrd_strain2 boards for 3000 ms in CAN2:1 of 1: missing eobrd_strain2 BOARD 10.0.1.3:CAN2:13
	4,917284	ERROR			from BOARD 10.0.1.3 (right_arm-eb3-j0_1) time=55s 433m 237u : CFG: CANdiscovery cannot find  1 missing eobrd_strain2 boards for 3000 ms in CAN2:1 of 1: missing eobrd_strain2 BOARD 10.0.1.3:CAN2:13

```

### Detailed context

Might be due to the cable connecting the FT and the board, but there is no evident damage and we don't have a replacement, so we cannot crimp it by ourselves this time. 

cc @DanielePucci

gabrielenava, 2024-02-06:
I have tested again the robot today: the FT sensor was detected this time and the robot started with no issues. I think this confirms that the problem is the cable that goes from the FT to the board. Probably the cable is a bit damaged but not broken yet, and sometimes fails depending on its position. Is it possible to replace it? It is completely accessible by removing the chest front cover, so fix it should not require to remove other pieces from the robot

gabrielenava, 2024-03-04:
this issue happened again today:

[image]

here the full logger:

[log_icub-head_yarprobotinterface[robotinterface]_22627.txt](https://github.com/robotology/icub-tech-support/files/14483031/log_icub-head_yarprobotinterface.robotinterface._22627.txt)

I have tried three times in a row, also trying things like restarting motors and even the switching off the power, but the issue keep occurring this time.

gabrielenava, 2024-03-04:
With @HosameldinMohamed we debugged this issue more in details:

1) there are two FT sensors connected to `right_arm_eb3-j0_1`. I disconnected them one by one discovering that the sensor that does not start is **the one inside the arm, not the jetpack one**

2) we opened the arm cover and with a tester we tested the cable: it seems ok. Then, we connected the FT sensor to a PC and checked it: also the FT was working fine. We mounted back the cover and, magically, both FTs were visible from the board. 

So I still think is a problem of the cable, probably something similar to what was happening in #1659: maybe it is a bit faulty, and when the cover presses it it does not work. When you open the cover and move it a bit, it restart working.

DanielePucci, 2024-03-05:
We are planning to run some experiments in the next days, so it would be nice if we might get some help for this, thanks 

CC
@sgiraz @Fabrizio69 @maggia80

AntonioAzocar, 2024-03-06:
Hi!
As you can see in the picture, there was the yellow wire damaged and that is why the **FT sensor** wasn't working properly.

[image]

Now I fixed it and with @AntonioConsilvio  we tested the FT sensor and it was **successful**!

We are waiting for feedback!

cc @sgiraz @gabrielenava

sgiraz, 2024-03-06:
Well done @AntonioAzocar! 🚀

I already discussed F2F w/ @gabrielenava!
The robot is working as expected.

Closing!

cc @DanielePucci

closed 2024-03-06 by sgiraz

DanielePucci, 2024-03-06:
Super thank you all!