robotology/mesh-support #1746: Short-circuited in chest

CescoWillemse, 2024-02-07:
### Robot Name 🤖

iCubTwente01 S/N:030

### Request/Failure description

The previous "caretaker" was testing the iCub before handing it to me. They could establish a network connection but no communication (no ping on 10.0.0.2). After attempting for approximately ten minutes, something in the chest short-circuited with a spark, a pop, and a little puff of smoke.

### Detailed context

Below a detailed report from the previous technician:

> After booting the laptop and turning on the 12V power supply the network connection did not start and I could not start a terminal session with icub. I tried several times. Each time a different set of LEDS inside the jaw of iCub turned blue but everytime also 1 or more turned red. I do not know the correlation of this and getting a proper connection, but it is just what I noticed.
>  
> While trying the N-th boot (I did not exactly count) I tried to gently push the shoulders more straight up to prevent a sudden bend-over forward. It was already straight, but I wanted to push it more straight. Therefore I pushed the chest-plate with a full hand slightly backwards. While doing this a ‘bang’ was heard and a small flash was seen and a thin trail of smoke came from behind the chest. Then I pushed the emergency button.

### How does it affect you?

I am the new lead researcher using this (notorious) iCub. I am comfortable with guided diagnosis of the exact issue but will have to (and be able to) delegate any physical repair jobs. In any case, let's start at the beginning and do some trouble shooting together. Thanks in advance!

sgiraz, 2024-03-11:
Hi @CescoWillemse, 

We have acknowledged your support ticket.
We are currently experiencing a high volume of support tickets and will be in touch as soon as possible with an update on how to proceed.

In the meantime, I would ask you to provide here some pictures of:
- the power supply configurations (Voltage and Current)
- a closeup of the boards around the head (with the cover removed) 

cc @AntonioConsilvio

AntonioConsilvio, 2024-04-19:
Hi @CescoWillemse, considering the problem described in this issue, it is very complex to provide instructions on how to deal with a probable short circuit on the iCub's main power supply. 
Also, the smoke and the fact that you can't reach the head could be symptoms of potential major damage. 

Therefore, to guarantee an appropriate level of support, I strongly recommend shipping the robot to us.
To organise the shipment of the robot, please contact @andreaderito who will provide you with all the necessary information.

CescoWillemse, 2024-04-22:
Thanks guys! I took the photos, but I hadn't had the time to upload them yet, plus they really didn't show anything. @andreaderito please drop me a line at c.willemse[at]utwente[dot]nl to organise shipping.