robotology/mesh-support #1747: Unable to detect firmware version after updating pc104 OS

MissingSignal, 2024-02-09:
### Robot Name 🤖

iCubGenova03 S/N:020

### Request/Failure description

Dear support,

we updated the OS (USB-live) of our iCub head from _debian Stretch_ to _Buster_.
We then installed (on the head) the latest version of superbuild running yarp 3.9.
When starting the robot through yarprobotinterface we face an error related to CAN firmware.
Apparently the old firmware is no more compatible and needs an upgrade.
We kindly ask your support to upgrade the firmware properly. 
So far from the [FirmwareUpdater](https://icub-tech-iit.github.io/documentation/icub_firmware/firmwareupdater/firmwareupdater/) tool we are not able to detect the current firmware.
 

### Detailed context

We attach the screenshots of the error when running yarp robot interface
[image]

sgiraz, 2024-02-12:
Hi @MissingSignal,

Can you describe the way you have built the `robotology-superbuild`?

Can you post here the output of the following commands from the `icub-head` ( pc104 board ):
- `df -h` command
- `ls -lah /usr/local/src/robot`
- `git status` within the /usr/local/src/robot/robotology-superbuild/build-pc104

>[!Note]
>The `build-pc104` folder should have been created (manually) during the robotology-superbuild installation and should contain the built files from `icub-head`.

MissingSignal, 2024-02-13:
Dear @sgiraz ,
I built the ```robotology-superbuild``` following this [guide](https://icub-tech-iit.github.io/documentation/sw_installation/icub_head_superbuild/).
In particular I compiled with the following flags:  

ROBOTOLOGY_USES_CFW2CAN          **ON** 
ROBOTOLOGY_ENABLE_ICUB_HEAD     **ON**
ROBOTOLOGY_USES_GAZEBO           **OFF**                                                                                                                                                                                                                                                                                   

Here I attach the output of the commands:

### df -h

```Filesystem      Size  Used Avail Use% Mounted on
udev            969M     0  969M   0% /dev
tmpfs           199M  3.1M  196M   2% /run
/dev/sda1       1.5G  1.2G  276M  81% /run/live/persistence/sda1
/dev/loop0      818M  818M     0 100% /run/live/rootfs/filesystem.squashfs
tmpfs           993M     0  993M   0% /run/live/overlay
/dev/sda2        13G  1.4G   11G  11% /run/live/persistence/sda2
overlay          13G  1.4G   11G  11% /
tmpfs           993M     0  993M   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           993M     0  993M   0% /sys/fs/cgroup
tmpfs           993M     0  993M   0% /tmp
tmpfs           199M     0  199M   0% /run/user/1000
```

### ls -lah /usr/local/src/robot
``` total 12K
drwxrwxrwx  1 root root 4.0K Feb  9 11:24 .
drwxr-xr-x  1 root root 4.0K Apr 21  2021 ..
drwxrwxrwx 13 root root 4.0K Feb  9 11:29 robotology-superbuild 
```

### git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   ../.ci/all-options-enabled-cache.cmake
	modified:   ../.ci/initial-cache.gh.cmake
	modified:   ../.ci/install_debian_and_script.sh
	modified:   ../.ci/script.sh
	modified:   ../.github/dependabot.yml
	modified:   ../.github/workflows/build-docker-images.yml
	modified:   ../.github/workflows/ci.yml
	modified:   ../.github/workflows/generate-conda-packages.yaml
	modified:   ../.github/workflows/matlab-one-line-install-test.yml
	modified:   ../.github/workflows/release.yml
	modified:   ../.github/workflows/update-latest-releases.yml
	modified:   ../.gitignore
	modified:   ../CHANGELOG.md
	modified:   ../CMakeLists.txt
	modified:   ../LGPL.TXT
	modified:   ../README.md
	modified:   ../apt-python.txt
	modified:   ../apt.txt
	modified:   ../cmake/BuildBayesFilters.cmake
	modified:   ../cmake/BuildBlockFactory.cmake
	modified:   ../cmake/BuildCppAD.cmake
	modified:   ../cmake/BuildGazeboYARPPlugins.cmake
	modified:   ../cmake/BuildHumanDynamicsEstimation.cmake
	modified:   ../cmake/BuildICUB.cmake
	modified:   ../cmake/BuildICUBcontrib.cmake
	modified:   ../cmake/BuildLieGroupControllers.cmake
	modified:   ../cmake/BuildOpenVR.cmake
	modified:   ../cmake/BuildOpenXR.cmake
[... 162 lines cut ...]
	modified:   ../releases/2022.02.0.yaml
	modified:   ../releases/2022.02.1.yaml
	modified:   ../releases/2022.02.2.yaml
	modified:   ../releases/2022.05.0.yaml
	modified:   ../releases/2022.05.1.yaml
	modified:   ../releases/2022.05.2.yaml
	modified:   ../releases/2022.08.0.yaml
	modified:   ../releases/2022.08.1.yaml
	modified:   ../releases/2022.09.0.yaml
	modified:   ../releases/2022.09.1.yaml
	modified:   ../releases/2022.11.0.yaml
	modified:   ../releases/2023.02.0.yaml
	modified:   ../releases/2023.02.1.yaml
	modified:   ../releases/2023.02.2.yaml
	modified:   ../releases/2023.02.3.yaml
	modified:   ../releases/2023.05.0.yaml
	modified:   ../releases/2023.05.1.yaml
	modified:   ../releases/2023.05.2.yaml
	modified:   ../releases/2023.08.0.yaml
	modified:   ../releases/2023.11.0.yaml
	modified:   ../releases/latest.releases.yaml
	modified:   ../scripts/install_robotology_packages.m
	modified:   ../scripts/robotologyUpdateLatestReleases.sh

no changes added to commit (use "git add" and/or "git commit -a")```

MissingSignal, 2024-02-13:
I think i found the error: 
we were not sourcing correctly the .bashrc_iCub file containing 
```export CFW2CANAPI_DIR=${DRIVER_ROOT}/cfw002/```  

I tried to recompile and now i get this error for CAN bus 1,2,3,4:
```[WARNING] Set baud rate not allowed from CanBusAccessPoint implementation
[INFO] using ICUB_CANMASKS_STRICT_FILTER option
[DEBUG] class 0 set
[DEBUG] class 1 set
[DEBUG] class 2 set
[DEBUG] class 3 set
[DEBUG] Can read/write buffers created, buffer size: 500
[INFO] CanBusResources::initialized correctly
[ERROR] recv buffer overrun ( 500 >=  500)
[ERROR] run()-pushReadMsg() failed on CAN bus 1
...
Registering new thread 0 out of 500

```

sgiraz, 2024-02-29:
Hi @MissingSignal,

See the comment in https://github.com/robotology/icub-tech-support/issues/1674#issuecomment-1970628402

sgiraz, 2024-03-05:
As of now, `iCubGenova03`:
- runs _debian Buster_ OS 
- use modern [robotology-superbuild v2024.02.00][1] instead of the old **separate** repositories installation
- has the `robotology-superbuild` installed inside the (16GB) USB live stick instead of mounting it from a dedicated desktop/laptop.

After discussing w/ @MissingSignal the pros and cons of having the `robotology-superbuild` installed within the (16GB) USB live stick, We agreed that on `iCubGenova03`, We try to go on w/ this desired configuration (similar to iCubHalfie01).  

Following up the activity reported in https://github.com/robotology/icub-tech-support/issues/1674#issuecomment-1979567657, I will add further support details in the next comments below.

cc @AntonioConsilvio @Fabrizio69 

[1]: https://github.com/robotology/robotology-superbuild/releases/tag/v2024.02.0

pattacini, 2024-03-06:
Relative PR:
- https://github.com/robotology/robots-configuration/pull/625

sgiraz, 2024-03-06:
Hi @pattacini,

I re-opened this issue because they probably need further support for this issue.

cc @AntonioConsilvio

MissingSignal, 2024-03-06:
Dear @sgiraz and @AntonioConsilvio  I think my last [PR](https://github.com/robotology/robots-configuration/pull/625) already solved the "wrong general calibration order of the robot's joints" you mentioned in [#1647](https://github.com/robotology/icub-tech-support/issues/1674#issuecomment-1979567657). 
For this reason I think we can close this issue 👍

Here I attach a video of the new calibration order. 

https://github.com/robotology/icub-tech-support/assets/36140565/eea39066-056d-43b7-8674-efa07ea174b8

sgiraz, 2024-03-07:
> Dear @sgiraz and @AntonioConsilvio I think my last [PR](https://github.com/robotology/robots-configuration/pull/625) already solved the "wrong general calibration order of the robot's joints" you [...]

Hi @MissingSignal,

I didn't catch it this fix! Thank you for your effort! 🙏🏻

Closing then.

cc @AntonioConsilvio

closed 2024-03-07 by sgiraz