robotology/mesh-support #1748: capacitor holder - 3D cad files request

julijenv, 2024-02-13:
### Robot Name 🤖

iCubTokyo01 S/N:022

### Request/Failure description

I would like to request the necessary files for reproducing the 3D pieces that hold the capacitor located in the back of iCub.
From my research it should be the file for reproducing the update **KIT_002 Update of capacitor support**.
Is it possible to request those files or should we order them?
We have a 3D printer in our lab. ORIGINAL PRUSA i3 MK3. The required files are .stl or obj. for the piece RC_TLR_003_P_060_00
Let me know,
best,
JJ

### Detailed context

The rear capacitor dont stick anymore to the pollax tape.

### How does it affect you?

Visible annoyance

AntonioConsilvio, 2024-02-13:
Hi @julijenv! you can download the capacitor holder file in .stl version here:

[rc_tlr_003_p_060_00_rev_a2.zip](https://github.com/robotology/icub-tech-support/files/14259720/rc_tlr_003_p_060_00_rev_a2.zip)

When you have printed the pieces, please write feedback if all goes well!

p.s 
Even if the words written in the issue description are less than those required by the github action, the issue will not be ignored! 
It is Just a reminder to avoid descriptions lacking information.

julijenv, 2024-02-14:
Dear @AntonioConsilvio ,
Just finished mouting 3 out of 4 capacitor holders 3D printed. Everything went smoothly.
Side note, the tie wrap is fine to be mounted on the rear capacitor because we can use the neck of the capacitor to hold it firmly into place, but the 2 capacitors on the front side of the robot are positioned way too close one another, so it is not possible to tie wrap the capacitor at the level of the neck, only the body is tied up. I reuse a bit a polax to prevent the capacitor to slide.
best,
JJ

closed 2024-02-14 by julijenv

Gandoo, 2024-02-26:
ciao @julijenv the capacitors on the front supports should be fixed like this, tighten the cable ties tightly. They shouldn't move because the capacitor leads hold them in place.

[image]

cc @AntonioConsilvio