robotology/mesh-support #1749: Loosen Neck

DanielePucci, 2024-02-13:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The neck seems to be loosen

### Detailed context

While the robot was walking, we saw some oscillatory motion of the robot head that were fishy. Visual inspection (see the video below) revealed that the neck is loosen 

### How does it affect you?

https://github.com/robotology/icub-tech-support/assets/5045846/7e9c6f8a-6164-4253-8f09-6e810ab37e35

fbiggi, 2024-02-14:
the Harmonic Drive screws have been tightened
and re-tensioned the tendons

@Fabrizio69 @maggia80 @AntonioConsilvio @sgiraz

closed 2024-02-14 by fbiggi

Fabrizio69, 2024-02-23:
This is ergoCub1.1 S/N:001, I change the title and the OP, so that this issue can be used for the fail/break report