robotology/mesh-support #1750: Loosen left thumb

DanielePucci, 2024-02-14:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The left thumb seems loosen.

### Detailed context

While going around the lab I started checking the robot and its hands and I discovered that the left hand thumb is loosen

### Additional context

https://github.com/robotology/icub-tech-support/assets/5045846/12166d8f-2c2d-4e9a-bd08-2d29061ddc6b

SimoneMic, 2024-02-15:
I think that this should be labeled as ergoCubSN001, isn't it?

DanielePucci, 2024-02-15:
Yes thanks, once we had also the colors in the robot name that helped avoid these mistakes

fbiggi, 2024-02-16:
yesterday I adjusted the screw on the engine

Fabrizio69, 2024-02-16:
...changed the name in the [OP](https://github.com/robotology/icub-tech-support/issues/1750#issue-2135042082) to avoid mistakes

sgiraz, 2024-02-16:
Thanks @fbiggi! 

We discuss how to prevent these issues internally, if possible..

Closing as solved.

cc @AntonioConsilvio

closed 2024-02-16 by sgiraz