robotology/mesh-support #1752: left-wrist fails to start or calibrate

vigisushrutha23, 2024-02-15:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

### Robot Name 🤖

ergoCub 1.0 S/N:001

### Request/Failure description

During an experiment involving bi-manual manipulation and navigation, we had to restart the robot due to `terminator` terminal crashing. When restarting the robot `left_wrist` throws 

```bash
 0.020122 <ERROR> from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=467s 663m 726u :  SYS: the EOtheInfoDispatcher could not accept a eOmn_info_properties_t item inside its transmitting queue. Number of lost items is  2.
```

Associated log is here 

[failed_comm_left_arm.txt](https://github.com/robotology/icub-tech-support/files/14301435/failed_comm_left_arm.txt)

cc @mebbaid @SimoneMic

### Detailed context

We tried to restart the whole setup (including the robot labtop `ergocub001-lap` and the issue persisted. We also tried running `yarpmotorgui --calib` but then we saw
[image]

[image]
[image]

### How does it affect you?

This affects https://github.com/ami-iit/lab-events-demos/issues/299

ale-git, 2024-02-16:
Is there an inversion in the connecting rod?

vigisushrutha23, 2024-02-16:
I am not sure exactly what you mean so I can't confidently say it. What I can say is that after we home the robot, the left wrist bends inwards to a great extent when compared to the right. We confirmed that the config files were correct so the wrist calibration seemed off before the fault occurred but worked fine for one experiment.

lornat75, 2024-02-19:
do we need a specific issue to track the process of fixing the wrist or this ticket is enough?

vigisushrutha23, 2024-02-21:
Closing the issue based on @sgiraz 's comment. Will reopen in case we face the issue again.

closed 2024-02-21 by vigisushrutha23