robotology/mesh-support #1754: Right hip roll do not calibrate

GiulioRomualdi, 2024-02-21:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Similar to https://github.com/robotology/icub-tech-support/issues/1753 we tried to switch on the robot and the right hip roll do not calibrate.

### Detailed context

The joint do not move at all and it remains in idle mode 

[image]

The encoders move if the joint is manually moved.

After calibration the robot appears like the following picture

[image]

### Additional context

Here the logger 
[log_ergocub-torso_yarprobotinterface_1440_right_hip_roll_failed.txt](https://github.com/robotology/icub-tech-support/files/14362045/log_ergocub-torso_yarprobotinterface_1440_right_hip_roll_failed.txt)

GiulioRomualdi, 2024-02-22:
We tried to increase the maximum PWM in the hip roll from 8000 to 12000 still it is not starting

AntonioConsilvio, 2024-02-23:
Hi! The joint didn't calibrate because the optical disk was stained due to the motor burn:

[image]

[image]

We will replace the optical disk, but We will not replace the motor, as although it has warmed up and has burnt areas, the phases have a resistance of 1.0 between red-black and white-black, and 1.1 between white-red, so they should be acceptable for the operation of the joint.

However, we do not know how long the joint could sustain prolonged use.

With @MSECode we will work on enabling the temperature reading.

cc @Fabrizio69 @fbiggi

AntonioConsilvio, 2024-02-27:
Hi @GiulioRomualdi, we mounted the rotor, calibrated and tested the joint. seems to be working correctly now.

However, during testing we detected a malfunction on the left hip roll.

We discovered that the motor was burnt out, so we proceeded with the replacement.

We have now tested the robot and both joints work correctly.

Here is the PR for the new joint calibrations:

- https://github.com/robotology/robots-configuration/pull/620

Please give us feedback as soon as possible, thank you!

cc @S-Dafarra

S-Dafarra, 2024-02-27:
The robot is able to walk. I noticed some more shaking, but they did not interfere too much. Also, I tested with the parameters we used in Bruxelles, which are more conservative than usual.

sgiraz, 2024-02-29:
Thank you for your feedback @S-Dafarra.

Closing!

closed 2024-02-29 by sgiraz