robotology/mesh-support #1755: All the joints friction seems to be increased

GiulioRomualdi, 2024-02-21:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Similar to https://github.com/robotology/icub-tech-support/issues/1753 and https://github.com/robotology/icub-tech-support/issues/1754 when unpacking the robot we noticed increased friction 
of some joints.

### Detailed context

In particular we noticed the friction increase of the following joints:
- right shoulder roll;
- both knees;
- right hip roll (with additional noise);

After some movement they seem a bit improved, in particular the shoulders.

sgiraz, 2024-02-21:
Hi @GiulioRomualdi,

Yep, maybe we have to perform some cleanups!

cc @AntonioConsilvio @fbiggi

sgiraz, 2024-03-15:
Hi @GiulioRomualdi,

We're going to clean the robot's box.
As of now, is the friction still present on the robot?

GiulioRomualdi, 2024-03-15:
The robot seems working as expected. Thank you!

closed 2024-03-21 by sgiraz