robotology/mesh-support #1756: Arm FTs cables damaged

S-Dafarra, 2024-02-22:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

We noticed that the cables of the FTs in both arms appear damaged. This causes the connection to be flaky and sometimes the ``yarprobotinterface`` does not start because the robot is not able to detect them

### Detailed context

See for example 
[image]

The cable seems to be damaged by the robot's own movement

We noticed this issue while being in Bruxelles for a demo

S-Dafarra, 2024-02-22:
> By the blue rubber cover I am wondering, is this ergoCubSN001 or the one in Rome?

Thanks for noticing. I made a mistake in selecting the robot from the dropdown menu

sgiraz, 2024-03-04:
Hi @S-Dafarra,

Do you have any news on this issue? Did the strain2 board disappear anymore?
If everything is ok, I will proceed to close this issue.

cc @AntonioConsilvio @SimoneMic

S-Dafarra, 2024-03-04:
> Hi @S-Dafarra,
> 
> Do you have any news on this issue? Did the strain2 board disappear anymore? If everything is ok, I will proceed to close this issue.
> 
> cc @AntonioConsilvio @SimoneMic

Hi @sgiraz, I haven't used the robot myself since we got back from the demo, and probably the arm FTs are still disabled at the moment.

sgiraz, 2024-03-14:
Hi @S-Dafarra,

Did you have the opportunity to use the FTs in the last days?

S-Dafarra, 2024-03-14:
They should have been enabled back, and we did not have any startup failure related to the FTs. I think we can close this, thanks!

closed 2024-03-14 by sgiraz