robotology/mesh-support #1757: Loosened right wrist

DanielePucci, 2024-02-27:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The right wrist seems loosened.

### Detailed context

While I was talking to @fiorisi saying that we need to overhaul the entire wrist-hand mechanism because of the difficulties in grasping a variety of objects with the ergoCub, I started touching the robot and I realised that the wight wrist seems loosen.

### Additional context

https://github.com/robotology/icub-tech-support/assets/5045846/c8bbda09-f789-46ac-9c81-3520bf934002

AntonioConsilvio, 2024-03-01:
Hi @DanielePucci, 
I have tensioned the screws and pinscrews shown in the photo below:

[image]

In fact you can see from the video that the bottom screw connecting the two wrist pinscrews was loose.

Also testing the pinscrews on the central shaft, they were loose.

Now that I have re-tightened everything, I ask for feedback if the situation has improved, thank you!

cc @S-Dafarra @GiulioRomualdi

sgiraz, 2024-03-06:
Well done @AntonioConsilvio!

@DanielePucci, If you agree, I would proceed w/ closing this issue.

sgiraz, 2024-03-15:
I guess everything is ok!
Closing.

cc @DanielePucci @GiulioRomualdi

closed 2024-03-15 by sgiraz

GiulioRomualdi, 2024-03-15:
> I guess everything is ok! Closing.
> 
> cc @DanielePucci @GiulioRomualdi

Yes as far as I know