robotology/mesh-support #1758: ergoCubHead computer not turning on and not reachable

SimoneMic, 2024-02-27:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The head computer on ergoCubSN001 is not reachable and the head display is not turning on. Probably the computer on the head is not powered.

### Detailed context

While setting up our experiments we realized that the head computer was not reachable, we also noticed that the head display wasn't turning on either. This is suggesting that the computer is not turning on.
So after three restarts of the whole robot and system, the same behavior re-occured.

I attach some pictures of the head:
[image]
[image]
[image]

cc @mebbaid @vigisushrutha23 

### How does it affect you?

We cannot proceed with our experiments, since we need to be able to access the lidar and camera

AntonioConsilvio, 2024-03-01:
Hi @SimoneMic, after some testing, we realised that the problem was that Xavier was not completing the boot sequence.

Thanks to @davidelasagna's help, he was able to recover by refleshing the Xavier's internal memory.

Together with @martinaxgloria we downloaded everything we needed (superbuild and the repositories to use lidar and RealSense + all the necessary dependencies).

We are having some problems with lidar (see [issue](https://github.com/robotology/icub-tech-support/issues/1705)), but everything else is working properly, see RealSense:

[image]

Thank you all for your help!

cc @maggia80 @sgiraz @Fabrizio69 @S-Dafarra

sgiraz, 2024-03-04:
Well done @AntonioConsilvio, @martinaxgloria, @davidelasagna!

Closing as solved!

closed 2024-03-04 by sgiraz