robotology/mesh-support #1759: Palm has snapped from wrist

dattilson, 2024-02-28:
### Robot Name 🤖

iCubHertfordshire01 S/N:009

### Request/Failure description

The plastic part has snapped off at the wrist, and I'm not sure how to remove the sensors/skin from the palm to access the screws to replace it.

### Detailed context

I'm unsure what caused it, but the 'neck' of the right palm, where it attaches to the motor has snapped, and so the part is in need of replacement.

I have managed to manufacture a replacement part using the creo part file as reference, however, I'm unfamiliar with the procedure for removing the palm's sensors/skin. In addition to 4 screws, it seems to be held in place by some kind of glue. This is unfortunate, as it needs to be removed to detach the palm.

Could you advise on the procedure for replacing the palm? Especially with regards to the removal (and reapplication, if necessary) of this glue.

### Additional context

[image]

sgiraz, 2024-02-29:
Hi @dattilson,

The broken part is made of carbon fiber.
I suggest replacing it with a brand new one that we can provide to you. In this case, contact @andreaderito ([andrea.derito@iit.it](mailto:andrea.derito@iit.it)) to proceed with the purchase process, please.

In the comment below my colleague @Elio75 will describe the procedure to replace the broken part.

cc @Fabrizio69 @AntonioConsilvio

Elio75, 2024-03-01:
CIAO @dattilson

First: 
- unplug the LP cable connector, which comes from the corn board, from the skin electronics. 

[image]

- remove the 4 side screws that fix the tissue to the palm, (be careful there is also double-sided tape between the tissue and the palm).

[image]

- At this point you will see three screws securing the palm to the hand, which you will have to remove to free it .
At this point you have two options:
- either try to reutilize the skin pcb, but know that The pcb being flexible will probably be damaged as it is attached to the palm by a double-sided tape , so it is very likely that you will break an internal track or be left with a component in your hand.
- In case you are interested in the palm skin working we can supply you with a new palm with the electronics integrated ready to be mounted on the hand.

cc @Fabrizio69 @AntonioConsilvio @sgiraz

dattilson, 2024-03-15:
We have attempted to recover the skin although we are unsure if we have recovered it undamaged. Would you be able to provide a quote for both options?

andreaderito, 2024-03-19:
Hi @dattilson please be so kind as to send me an email at andrea.derito@iit.it and I will be pleased to provide you with details - please indicate in your email the precise delivery address of the goods, 1 (or also 2) reference person(s) at delivery address (name, surname, email address, and important - a telephone number, possibly a mobile number) - I ask you to send such data via email to me and not here in this issue for privacy reasons, thank you for understanding and cooperating - best
cc @Fabrizio69 @AntonioConsilvio @sgiraz

andreaderito, 2024-04-03:
Hi @dattilson up to date I did not receive any email from your side (see my comment [above](https://github.com/robotology/icub-tech-support/issues/1759#issuecomment-2007844760)). I was wondering whether you have been successful to re-mount the parts and they work, so - maybe - you do not need the spares any longer? Looking forward to your kind feedback.

Fabrizio69, 2024-04-30:
Hi @andreaderito, here the parcel dimensions and weight:
- 17x17x20 cm
- 0.5 kg

cc: @sgiraz @AntonioConsilvio @Miche19 @maggia80

andreaderito, 2024-05-02:
Dear @dattilson I'm going to contact you via email for further information so we may proceed - pls keep ready to react to my email, best

AntonioConsilvio, 2024-12-18:
The palm has been sent and receipt confirmed.

I am closing the issue!

closed 2024-12-18 by AntonioConsilvio