robotology/mesh-support #1760: Fault button broken

gabrielenava, 2024-02-28:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The robot fault button does not work anymore

### Detailed context

I was trying to start the robot, but it keeps failing with error "external fault button pressed". When I switched to another fault, the robot started with no issued

### Additional context

Here is the faulty fault, in the robot kitchen close to the AMI area.

[image]

AntonioConsilvio, 2024-04-24:
Hi @gabrielenava, I tested the fault button and it seems to work properly.

https://github.com/robotology/icub-tech-support/assets/114915464/a82719ad-11cc-4858-aff2-16e17185b3aa

Maybe you need to press harder when inserting!

AntonioConsilvio, 2024-05-22:
Hi @gabrielenava! Have you tested if inserting the connector a little harder works properly?

gabrielenava, 2024-05-23:
hello! we did it today. First time the robot started with no issues. Second time we had issues at startup, similar to what reported [above](https://github.com/robotology/icub-tech-support/issues/1760#issue-2159253865). But at this point I do not know if the problem is the fault button or it is yet another declination of https://github.com/robotology/icub-tech-support/issues/1799

if you prefer we can close this issue and keep https://github.com/robotology/icub-tech-support/issues/1799 open to investigate these startup issues.

AntonioConsilvio, 2024-05-24:
Thank you for the feedback @gabrielenava! Closing!

closed 2024-05-24 by AntonioConsilvio