robotology/mesh-support #1763: Right hip roll motor overheating

evelyd, 2024-03-01:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The right hip roll motor overheated. 

### Detailed context

It failed a couple of times in overcurrent, then by overheating, reaching over 100 degrees. This happened before it was supposed to start walking (i.e. had not moved yet). Icub Tech checked with a thermal camera and determined that the motor is probably burnt and needs to be replaced. 

### Additional context

[image]
[image]

[image]

https://github.com/robotology/icub-tech-support/assets/17877131/8734270f-ba8d-4ea2-b80b-2c5c2c01036b

### How does it affect you?

Walking tasks can't be tested while there is this issue.

GiulioRomualdi, 2024-03-01:
cc @DanielePucci @S-Dafarra @isorrentino

sgiraz, 2024-03-04:
Hi guys,

As agreed, in the next few days @fbiggi @AntonioConsilvio are going to clean/replace the motor.

@fbiggi, during the intervention to solve this issue, consider also #1755 if possible, please.

sgiraz, 2024-03-05:
Today @fbiggi replaced the motor and with the help of @MSECode recalibrated the joint.

Let us know if the intervention resolved [the issue mentioned above][1].

cc @evelyd @GiulioRomualdi @S-Dafarra 

[1]: https://github.com/robotology/icub-tech-support/issues/1763#issue-2163502824

vigisushrutha23, 2024-03-06:
Hi @sgiraz. Yesterday, @mebbaid, Simone and myself ran multiple walking experiments. We did not face the hip roll motor overheating issue through 6 experiment runs, involving walking a distance of 1.5m for four of them while performing manipulation. We encountered the ankle motor overheating but restarting the motors after a brief cooldown period allowed us to do one more experiment without any issue.

MSECode, 2024-03-06:
Good, good to know and thanks for the feedback regarding the motor behaviour. 
Thanks to @AntonioAzocar for the support on the cabling and calibration too. 

cc: @valegagge

sgiraz, 2024-03-06:
Awesome work, everyone! 🚀
Closing.

closed 2024-03-06 by sgiraz