robotology/mesh-support #1768: Neck joint broken

evelyd, 2024-03-08:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The neck roll joint cable broke.

### Detailed context

During a walking task, the neck suddenly broke. On closer inspection the cable snapped completely. The type of motion didn't require the neck to move very much.

### Additional context

[image]

https://github.com/robotology/icub-tech-support/assets/17877131/249efdbb-642c-4402-bd5c-5fb919305815

[robot_logger_device_2024_03_08_13_09_54.md](https://github.com/robotology/icub-tech-support/files/14538048/robot_logger_device_2024_03_08_13_09_54.md)

Fabrizio69, 2024-03-08:
I think that this tendon was almost broken before the experiment you did...

AntonioConsilvio, 2024-03-08:
Hi @evelyd, I replaced the broken tendon with e new one and tested the robot, it works as expected! Closing!

cc @sgiraz

closed 2024-03-08 by AntonioConsilvio