robotology/mesh-support #1769: Backlash in the right shoulder arm support

gabrielenava, 2024-03-08:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

There is backlash in the support of the right shoulder joints.

### Detailed context

When the robot arm is rigid (position control) and a person tries to move the shoulder joints, we can see a backlash between the shoulder and its support (in the figure below). This backlash is only on the right shoulder, while on the left shoulder is not present (maybe because of the intervention done in https://github.com/robotology/icub-tech-support/issues/1692

[image]

I have tried to screw the accessible screws and the situation improved. I think it might be just a matter of screwing the remaining screws, but they are not accessible. We need to unmount part of the arm to do it, so I open this issue for asking support.

cc @DanielePucci 

### How does it affect you?

It affects iRonCub experiments.

gabrielenava, 2024-04-25:
thank you @AntonioConsilvio!

AntonioConsilvio, 2024-05-07:
Closing for this comment:

- https://github.com/robotology/icub-tech-support/issues/1792#issuecomment-2090763510

closed 2024-05-07 by AntonioConsilvio