robotology/mesh-support #1770: Right arm power cable is damaged

gabrielenava, 2024-03-08:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

A power cable in the upper right arm is damaged.

### Detailed context

While fixing the issue https://github.com/robotology/icub-tech-support/issues/1745 with @AntonioAzocar, we noted that there is a damaged power cable inside the right upper arm:

[image]

We temporary fixed it with some black tape, but it would be better to replace it. The cables had already other fixes, so I think it is quite old.

cc @DanielePucci 

### How does it affect you?

Might be a problem if the wire is exposed and gets in contact with the upper arm boards or metal support.

AntonioAzocar, 2024-04-23:
Hi!
In addition to the cables mentioned in the main comment, there were also some damaged ones, so I re-did them all.

[image]

cc @gabrielenava

AntonioConsilvio, 2024-04-24:
Good job @AntonioAzocar! 🚀