robotology/mesh-support #1771: [Thumb distal broken]

ddetommaso, 2024-03-08:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

The tendon of the thumb need to be replaced

### Detailed context

-

### Additional context

[yarprunlog_08_03_2024_15_13_55.txt](https://github.com/robotology/icub-tech-support/files/14539159/yarprunlog_08_03_2024_15_13_55.txt)
[image]

### How does it affect you?

High impact considering the possibility of delaying data collection in clinical settings.

sgiraz, 2024-03-13:
Ok @ddetommaso,

As agreed the maintenance activity is scheduled for tomorrow morning on-site (Boggiano Pico).
@AntonioConsilvio and @fbiggi will be in place to fix the thumb

cc @Fabrizio69

sgiraz, 2024-03-14:
Well done guys! I think we can close this issue!

cc @ddetommaso

closed 2024-03-14 by sgiraz