robotology/mesh-support #1772: Mecanical issues on the left hand

ddetommaso, 2024-03-08:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During calibration the left hand get stuck with the fingers closed. Probably there is some issue with the tendons of the thumb

### Detailed context

-

### Additional context

https://github.com/robotology/icub-tech-support/assets/3440640/eddfdc39-6c17-4b04-b759-318128576741

### How does it affect you?

Medium impact. Data collection with participant should start in 2 weeks.

sgiraz, 2024-03-11:
Hi @ddetommaso,

We suppose that it is a wiring issue. Can you provide a log of the `yarprobotinterface` and a screenshot of the `yarpmotorgui` after the calibration phase, please?

cc @AntonioConsilvio

ddetommaso, 2024-03-11:
Hi @sgiraz,
please attached you can find more details about the issue.

[log_icub-head_yarprobotinterface_1843.txt](https://github.com/robotology/icub-tech-support/files/14557462/log_icub-head_yarprobotinterface_1843.txt)

[image]

AntonioConsilvio, 2024-05-20:
Hi @ddetommaso, during this intervention:

- https://github.com/robotology/icub-tech-support/issues/1805

We have also checked the problem mentioned in this issue. All fingers calibrate properly now ! Closing!

closed 2024-05-20 by AntonioConsilvio