robotology/mesh-support #1774: Fingers cannot reach lower limit

PasMarra, 2024-03-11:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While commanding fingers closure, using the motorgui, I've noticed that some fingers got stuck before reaching the set point.

### Detailed context

# right_thumb

`right_thumb_oc` get stuck at 52 deg.

[image]
[image]

# right_fingers

As you can see, we are missing something on the other fingers.

[image]
[image]

# left_thumb

`left_thumb_oc` works fine.

[image]
[image]

# left_fingers

As you can see, we are missing something on the other fingers.

[image]
[image]

### How does it affect you?

Cannot perform any grasp

simeonedussoni, 2024-03-12:
today i gave a look at this problem. 

It turned out to be due to the loose tensioning of the related tendons.

most of the issue could be solved acting on the tensioner but for the right middle finger this was not sufficient. 
Actually the tensioner arrived at the end and exited from the threaded receptacle. 

I prepared a small spacer to be installed to temporarily fix this.

meanwhile I also finely-tuned all the fingers in order to reach endstops in closure while preserving the opening range of motion.

It turned out that the endstop need to be finely calibrated also in software since the FAP encoder read in some cases a few degrees less tha 90 with the finger fully pushed to the endstop (but this is not a problem from the point of view of the grasping at least at the present development stage)

cc @sgiraz @AntonioConsilvio @Fabrizio69 @fbiggi

AntonioConsilvio, 2024-03-21:
Hi @PasMarra, me and @fbiggi intervened on the middle finger of the right hand.

The finger was getting stuck in the middle of its range of motion, and we disassembled the finger and replaced the spring that had stretched. On opening the finger, we found that the fingertip connector had detached and stuck  in the finger leverage, compromising the finger closure.

[image]

I also replaced the FAP cable as it was damaged @fgarini.

[image]

Now the robot can be use, so please give us a feedback about the functioning of fingers! Thank you!

cc @simeonedussoni

AntonioConsilvio, 2025-01-27:
Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/1819#issuecomment-2616180058

closed 2025-01-27 by AntonioConsilvio