robotology/mesh-support #1778: Left wrist loosened

DanielePucci, 2024-03-19:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

I noticed that the left wrist is loosened.

### Detailed context

I was seeing that the left wrist was moving in a strange way. For this reason, I started touching the left wrist that seemed loosened.

### Additional context

https://github.com/robotology/icub-tech-support/assets/5045846/fa76f9af-7813-45e8-ad1a-7436e2545377

AntonioConsilvio, 2024-03-21:
Hi @DanielePucci, today I put the belts that were about to come off the pulley back in place (see [issue](https://github.com/robotology/icub-tech-support/issues/1728)), also tightened some of the pin screws that were loose and some calibrated screws that hold the wrist linkages (similar to this [issue](https://github.com/robotology/icub-tech-support/issues/1757)). 

The situation seems better, but I am asking for feedback on this. Thanks!

cc @Fabrizio69 @fiorisi

DanielePucci, 2024-03-25:
Seems better @AntonioConsilvio !

AntonioConsilvio, 2024-03-26:
Thank you for the feedback @DanielePucci! Closing!

closed 2024-03-26 by AntonioConsilvio