robotology/mesh-support #1779: Cannot read right hand tactile sensors

PasMarra, 2024-03-19:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

I'm not able to read measurements from the xela tactile sensors mounted on the right hand of the robot.

### Detailed context

Yesterday I tried to read the measurements from the xela tactile sensors mounted on the right hand of the robot but I wasn't able to establish any communication with them. Given that, while I was there, I've also tried the exact same procedure with another set of sensors and it worked, I suppose that something is wrong with the wiring.

cc @randaz81

maggia80, 2024-03-22:
@PasMarra do you discover with FW updater or any other tool the MTB4 board?

PasMarra, 2024-03-22:
Hi @maggia80, I don't need to use the FW updater in order to communicate with the sensors. What i do is to physically connect my pc to the xela sensors by means of the external cable that now is hanging from the forearm. Then, i follow [this](https://github.com/hsp-iit/xela-tactile-sensors-ergocub/tree/main?tab=readme-ov-file#information_source-how-to-establish-can-connectivity) guide to establish CAN connectivity.

lornat75, 2024-04-08:
@maggia80 can some of support check the wiring? maybe it is something interrupted

AntonioConsilvio, 2024-05-20:
Hi @PasMarra, fingertips should be connected properly now.

Please, give us feedback as soon as possible, thank you!

PasMarra, 2024-05-21:
Hi @AntonioConsilvio, before showing you the test result I would like also to point out another issue that often arises, even this morning, that also affects the communication: the cable connecting the sensor to the board is often disconnected, I don't know if this happens because the cable is not long enough or the connection between the cable and the sensor is loose.

Anyway, this is the can messages I receive.

[image]

The message ID `XY` is to be interpreted this way: `X = finger`; `Y = taxel`; with `thumb = 0`, `index = 1` and so on.
As you can see, the communication happens flawlessly with the thumb (red), index(green) and middle finger (purple), as I receive data from all the 7 taxels, while I only partially receive data from the ring finger and I cannot read at all any data from the pinky.

I've performed these tests:
1- switch ring finger's sensor with the pinky's one: the situation does not change, meaning that the I still cannot read pinky's sensor, while I only receive the taxels 0,1,2 from the ring finger's sensor;
2- test the ring finger and pinky's sensors on the bench setup: I can read all the taxels for both of them;
3 - switch the thumb and index finger's sensors respectively with the pinky's and the ring finger's ones: again, the situation does not change;

AntonioConsilvio, 2024-05-21:
cc @simeonedussoni

simeonedussoni, 2024-05-21:
ciao @PasMarra 

how did you made the switch mentioned in your test? you exchanged the cables on the MMA? and the OFF channels remained at the address 3XX and 4XX? and how did you made test number 2? you connected another set of fingertips or the same in the hand?

PasMarra, 2024-05-21:
> ciao @PasMarra
> 
> how did you made the switch mentioned in your test? you exchanged the cables on the MMA? and the OFF channels remained at the address 3XX and 4XX? and how did you made test number 2? you connected another set of fingertips or the same in the hand?

Hi @simeonedussoni,
test 1 and 3: for each sensor, I've disconnected the cable on the sensor's side and physically switched the sensor position on the hand. Basically the circuit has not been changed, just the sensor position;
test 2: I've unmounted the sensors from the hand and I've connected them on a bench setup, just to check that all the taxels were really working.

simeonedussoni, 2024-05-21:
> Hi @simeonedussoni, test 1 and 3: for each sensor, I've disconnected the cable on the sensor's side and physically switched the sensor position on the hand. Basically the circuit has not been [...]

Ok, so this point to a problem in the cables. I'll going to check it.

AntonioConsilvio, 2024-08-07:
> Ok, so this point to a problem in the cables. I'll going to check it.

Hi @simeonedussoni! Have you had a chance to take a look?

simeonedussoni, 2024-08-07:
ciao @AntonioConsilvio 
as I'm finalising the new FW I'd work on this problem once finished to have a single testing of both.

simeonedussoni, 2024-08-23:
Actually I reconnected the cables on the **left** hand. I suppose there is also to reconnect the cables in the **right** hand. 
We know the origin of the detachment and already applied the necessary corrections to the next version of the phalanges.