robotology/mesh-support #1781: Neck motor tendon broken

evelyd, 2024-03-21:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During tests where the head was not explicitly controlled, using the WalkingModule, the neck jerked at a certain point and broke.

### Detailed context

The walking motion was smooth up until the point that the neck jerked, and the controller we were using was not sending desired neck positions.

### Additional context

Error message received when it broke:
[ERROR] from BOARD 10.0.1.20 (head-eb20-j0_1) time=4871s 531m 342u :  MC: i2t limit exceeded. The motor has been turned off to prevent it from being damaged by overheating due to a continuous high current. (Joint=neck_pitch (NIB=0))

Repository branches in use on the ergocub-torso at time of failure:
robots-configuration: devel
walking-controllers: dnn-mcp-prepare

https://github.com/robotology/icub-tech-support/assets/17877131/21deb219-a46b-4843-900e-e0f7f546ede9

Plot of the measured neck joint positions:
[image]

SimoneMic, 2024-03-21:
Today on Friday 22 we have a small demonstration at around 11.30/12.00.
If the neck could be repaired in the morning it would be really super ❤️

I have switched the booking slot with Ines', so in the morning is free and if someone needs it for repairing it's available.

Thanks!

AntonioConsilvio, 2024-03-22:
Hi @SimoneMic, We plan to intervene before your Demo!

AntonioConsilvio, 2024-03-22:
@fbiggi and I replaced the neck tendon and the robot has been tested! 

cc @SimoneMic

AntonioConsilvio, 2024-03-22:
For some information about the tendon rupture see this [comment](https://github.com/robotology/icub-tech-support/issues/1773#issuecomment-2014727352)

Closing!

closed 2024-03-22 by AntonioConsilvio