robotology/mesh-support #1782: missing screws on new face - servo motor

julijenv, 2024-03-25:
### Robot Name 🤖

iCubTokyo01 S/N:022

### Request/Failure description

On our freshly received new face, the rotating arm attached to the servo motor is detached. Apparently the screw is missing.

### Detailed context

After some time operating the blinking start to do not behave as intended. I could hear the servo spinning but not the eyelids.
Will it be possible to get the missing screw?
I also noticed that the screws mounting on the servo motor and branched activating the eylid are hexagonal, from what I recall, they should be crossed.
can I get all the information for those screws in order to buy them if you cant provide them to me.
I would e also happy to receive information on how to recalibrate the eyelid when I will be remounting everything.
let me know,
best

### Additional context

[image]
[image]

### How does it affect you?

I reattached the old head meanwhile as we are currently showcasing the robot.

AntonioConsilvio, 2024-03-26:
Hi @julijenv, the screw that holds the leverage mechanism to the motor shaft has unscrewed. The screw is a cross screw M1.6x4.

The other 'hexagonal' screws are Torx screws, in case you have the torx screwdrivers you can disassemble and reassemble them without any problems, or replace them with cross screws if you prefer. (Actually, you should be able to unscrew torx screws even with a cross screwdriver in case you don't have the torx screwdriver).

However, the missing screw has the parameters mentioned above: cross M1.6x4.

[image]

julijenv, 2024-03-28:
Hi @AntonioConsilvio ,
can you tell me if the thread distance is American or European, from what I recall it is an American screw, can you take a picture of the normal M1.6x4 next to that black screw or confirm they are the same.
thanks in advance

AntonioConsilvio, 2024-03-28:
Hi @julijenv, You may be right, the thread should be American. 
However, we tried fitting an M1.6x4 metric instead of the official black screw on a servomotor, and it fit without any problems!

[image]

So in case you are only equipped with metric screws, you can try using those. 
But yes, the official screw should have the American thread.

julijenv, 2024-04-01:
Dear @AntonioConsilvio ,
I succeeded in reput back in place everything, thanks for the overall details,
best,
Julien

closed 2024-04-01 by julijenv