robotology/mesh-support #1789: Right front foot F/T streams constant values

GiulioRomualdi, 2024-04-03:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

This morning we had a demo and we noticed that the robot was not able to walk anymore.

After the demo we log a dataset and we check the values of the FT. Right front foot F/T streams constant values

### Detailed context

[image]

### Additional context

I asked @G-Cervettini to check the dataset he collected yesterday and we just noticed that in the dataset collected at 19:58 the sensor was still working. The first experiment where the sensor stops working is at 20:06. :sob:

cc @LoreMoretti @G-Cervettini @carloscp3009 @danie

### How does it affect you?

We cannot make the robot walk

AntonioConsilvio, 2024-04-03:
Hi @GiulioRomualdi, I would exclude problems with the wiring and CAN communication for two reasons, the first is that the `yarprobotinterface` starts, and the second is because strain2 is visible from `FirmwareUpdater`.

Below you can see the values transmitted by the FT channels:

[FT.webm](https://github.com/robotology/icub-tech-support/assets/114915464/fc147266-3e8e-4c06-a7fe-1f2324945c7a)

And the the values of the forces, that are always constant and are apparently meaningless:

[image]

I think the cause of the problem is the strain2 itself and it needs to be replace. 

cc @maggia80 @Fabrizio69

AntonioConsilvio, 2024-04-03:
@fbiggi and I replaced the damaged FT sensors (code: 12487.A3 SN: 116) with a new one (code: 12487.A5 SN: 128).

The robot has been tested with @GiulioRomualdi and it works properly now (it can walks now)!

However, since the intervention was quite invasive, I wait a couple of days before closing the issue.

cc @maggia80 @Fabrizio69

GiulioRomualdi, 2024-04-03:
Thank you again @AntonioConsilvio and @fbiggi !

GiulioRomualdi, 2024-04-04:
Today, we also tried to run a walking experiment with a new walking architecture. We didn't experience any problem related to the FT

AntonioConsilvio, 2024-04-11:
Thank you for the feedback!

closed 2024-04-11 by AntonioConsilvio