robotology/mesh-support #1790: Right leg ankle roll hardware fault due to overheating at startup

SimoneMic, 2024-04-03:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

`r_ankle_roll` goes into HW fault for overheating when launching the yarprobotinterface.
I have tried turning on and off the motors and restarting it four times, but the error occurred nonetheless.
I am attaching the pictures below in detailed context.

### Detailed context

`robots-configurations` branch is in devel.
The usual error that occurs is: 
```
[ERROR] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=283s 433m 172u :  src LOCAL, adr 0,(code 0x02000012, par16 0x0001 par64 0xfffffffffffffc8a) -> MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. In par16 = ID of the joint. In par64 0xFFFF is the mask of the raw tmperature feedback. .
```
I attach also the full log
[overheating_error.txt](https://github.com/robotology/icub-tech-support/files/14853973/overheating_error.txt)
[image]

Robot fotos in additional context

### Additional context

[image]
[image]
[image]

GiulioRomualdi, 2024-04-03:
cc @MSECode and @valegagge

valegagge, 2024-04-04:
Hi @SimoneMic,
did the error occur after or during the calibration?
Thanks

SimoneMic, 2024-04-04:
During the calibration

AntonioConsilvio, 2024-05-13:
Hi! This problem was due a damaged cable that connect the TDB sensor to the 2FOC board. We fixed the cable and it works properly now!

closed 2024-05-13 by AntonioConsilvio