robotology/mesh-support #1791: Help with error of 10.0.x.x boards

Ryo-Naohara, 2024-04-03:
### Robot Name 🤖

iCubLisboa01 S/N:004

### Request/Failure description

When starting icub's yarprobotinterface, it won't move and gives error message

### Detailed context

The error it gives seem related to the CAN boards connecting to 10.0.10.1, we are not sure on what to do.

[new_log_icub-head_yarprobotinterface_1403.txt](https://github.com/robotology/icub-tech-support/files/14855370/new_log_icub-head_yarprobotinterface_1403.txt)

@plinioMoreno

plinioMoreno, 2024-04-17:
Hi @AntonioConsilvio,

I want to add more information on this issue:

Regarding the left arm error:

```
11.415500 <ERROR>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src CAN2, adr 0, time 58s 713m 128u: (code 0x00000019, par16 0x4001 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
```
We run the `FirmwareUpdater` and the software was not able to find the CAN2 board that is connected to the ETH board 10.0.0.1. Is this a communication problem that we can check with the cables? Or how can we find out what is the source of this problem?

Regarding the right arm error:
```
4.627937 <ERROR>  from BOARD 10.0.1.29 (right_arm-eb29-j12_15), src LOCAL, adr 0, time 52s 149m 57u: (code 0x04000002, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag02 + mais timeout
```
We also run the `FirmwareUpdater` and the software was not able to find the board that is connected to the MAIS board. How can we find out what is the source of this problem?

Looking forward to hearing from you,

AntonioConsilvio, 2024-04-18:
Hi @plinioMoreno @Ryo-Naohara, Sorry for the late reply!

There is most probably a problem with the wiring, in fact on CAN2 of the `eb1` (`10.0.1.1`) board there should be a `strain`. It is possible that the cable has been damaged. 
The `strain` is in the left shoulder (red 🔴):

[image]

While the wiring is as follows, the `strain` (red 🔴), starts a cable (green 🟢) which runs to a male-female connection (yellow 🟡), and then starts again and runs into the sheath (green 🟢) and arrives at the EMS eb1 (10.0.1.1 - blue 🔵).

[image]

I recommend checking the wiring and testing the continuity with the tester. You can also test that there are no short circuits between the cables.

As for the right arm, I had given some indication in this comment:

- https://github.com/robotology/icub-tech-support/issues/1678#issuecomment-1938425288

Please, send us feedback and if you have any questions feel free to ask!

Ryo-Naohara, 2024-04-23:
cc: @plinioMoreno 

I tried checking the cable circling in light blue its has a yellow tag that says 1n2, and at least from the board pins up to the connector there weren't any faults or short circuits.
On icub Firmware updater we can see all CAN boards but the  one mentioned has "strain" under the board 10.0.1.1

What else can be done to try and fix this?

[image]

plinioMoreno, 2024-05-09:
Hi @AntonioConsilvio,

@Ryo-Naohara checked the cables from the control board (blue) to the male-female plug (yellow). The missing part to be checked is from the male plug to the encoder (`strain`). To be sure that the problem on that last part, @Ryo-Naohara did another cable that replaces the cable from the control board to the male-female plug. He run the `firmwareUpdater` and the `strain` was not found. Thus, the problem is with the cable that connects male-female plug with the `strain`. Now we need some help on how to check the cable from the `strain` to the yellow plug. (Do we have to unmount the arm part to check the strain?)

Let us know how to proceed.

AntonioConsilvio, 2024-05-13:
Hi @plinioMoreno @Ryo-Naohara, we are up to date on your issue, unfortunately due to several emergencies we are unable to offer an accurate diagnosis of your problem. We will update you as soon as possible.

We apologise for the waiting time.

AntonioConsilvio, 2024-05-23:
Hi @Ryo-Naohara @plinioMoreno! 

Since removing the `strain` sensor is a challenging and invasive operation, we have to be sure that the sensor is actually the problem.

So first I would ask you to perform the following tests:

- Where the `strain` cable starts a led should be visible when the sensor is turned on. So with the robot switched on I would try to see if this led is visible or not.

  https://github.com/robotology/icub-tech-support/assets/114915464/ffa5debb-154f-43b3-8263-7c9fb7890533

- You can try testing the crimping of the cable with tweezers, as in this video (if you have not already done):

    https://github.com/robotology/icub-tech-support/assets/114915464/801adf61-1312-4a0f-99fa-3175fac4bf6a

  You can try pulling the cable with tweezers, without tearing it, just to see if the crimp still holds.

- Try disconnecting the male-female (yellow 🟡 in the previous photos) and connect the `strain` cable to the MC4-plus `EB24` (`10.0.1.24` - the board closest to the mechanics of the 3).

  [image]

 
  You should see a CAN cable connected to the `MTB` boards (i.e. the skin boards mounted on the cover - cable name: 2N2) that should start from the `EB24`. 
You can temporarily disconnect the 2N2 cable from the `MTB` and connect it to the `strain` (matching the red - green - white - black colors) and try to see if the `strain` board is visible from `FirmwareUpdater` under board `10.0.1.24`.

> [!Note]
> To perform this test, the female connector of the `strain` must be 6-pin, and 5V and not less must be coming from the MC4-PLUS (it can be tested with the multimeter).

For any doubts about wiring, I recommend consulting the wiring diagrams:

- https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2.5/iCub%20wiring%20Rev3.3%20COM-EXP%20and%20MC4-PLUS%20and%20BatteryPack/pdf/iCub2.5-E3.3.2_5672_Logic.pdf

Please, send us feedback and if you have any questions feel free to ask!

Ryo-Naohara, 2024-05-23:
Thank you very much, i'll reply with updates as soon as possible

Ryo-Naohara, 2024-05-24:
Hello, i've tried to see the leds blinking but it seems they were off on both arms, even though in the right arm we can find the strain, and on the left one we can't, i think we still have the older version of the shoulders, and only the parts below the elbow are the new ones, so that might be why they might not be blinking.

For the pins i've tried pulling on them with a tweezer to see if any came loose but none did, the connector itself seems to have been glue with resin.
[image]

I've also tried doing what you recommended with putting the strain cable into where 2N2 was connected but couldn't find the mais board inside board 24.
[image]

We are going to try to remake the strain cable connector to see if inside the resign there was some wire that broke but won't come out when pulled because of the resin.

AntonioConsilvio, 2024-06-18:
Hi @Ryo-Naohara @plinioMoreno, regarding this issue, at this moment we do not have any `strain` sensors to send you for replacement.

We will update you as soon as we have new sensors!

plinioMoreno, 2024-07-17:
Hi @ AntonioConsilvio,

Do you have an idea on when you are expecting to build more `strain` sensors? Just for us to try to plan the timing with the students that are using the robot?

AntonioConsilvio, 2024-07-18:
Hi @plinioMoreno, we will provide you more info very soon.

AntonioConsilvio, 2024-07-19:
Hi @plinioMoreno! Good news. We have a sensor available for shipment.

For all shipping details the contact is @andreaderito.

You will probably be contacted by @andreaderito next week!

plinioMoreno, 2024-07-22:
Hi @AntonioConsilvio, those are great news!

And about the toolkit that helps for debugging hardware/electrical problems, can you send that as well?