robotology/mesh-support #1792: Check motors and transmission elements of shoulder and elbow of iRonCub3

gabrielenava, 2024-04-14:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

We would like to check the status of the robot arms motors after the experiments of Friday.

### Detailed context

During the experiments of Friday, the robot arms undergo a huge stress and I am worried about the status of the motors and of the transmission elements. In particular, I have tried to move the arms while position controlled and I saw that there is, qualitatively, a bit more backlash than before. 

We would need that a person more expert than us checks the arms' motors status, screw the right screws (there are many and we don't know which screw to touch and how much they should be tighten) and verify that the transmission elements like the bearings are behaving normally. 

cc @danie

### How does it affect you?

It compromises safety of the experiments.

AntonioConsilvio, 2024-04-24:
Hi @gabrielenava! I carried out a check of the tightening of the screws and the robot wiring. I focused more intensively on the upperbody.

I checked the Ethernet wiring and rewired some damaged cables.

I ask for feedback when you are able to test the robot! Thank you!

gabrielenava, 2024-04-25:
thank you @AntonioConsilvio!

gabrielenava, 2024-05-02:
>I ask for feedback when you are able to test the robot! Thank you!

we turned on the robot and tested it with jets off. Everything seems working fine! there was a problem at startup: the hip joints did not calibrated, even after restarting the robot multiple times. In the end, we calibrated them using the morotgui. We opened a dedicated issue for this problem: [#iCubGenova0](https://github.com/robotology/icub-tech-support/issues/1799)

the rest seems ok! thank you!

AntonioConsilvio, 2024-05-07:
> we turned on the robot and tested it with jets off. Everything seems working fine! But there was a problem at startup: the hip joints did not calibrate, even after restarting the robot multiple times. In the end, we calibrated them using the morotgui. We opened a dedicated issue for this problem: [#iCubGenova0](https://github.com/robotology/icub-tech-support/issues/1799)

Since you have created a dedicated issue, I will proceed with the closing of this issue! Thank you!

closed 2024-05-07 by AntonioConsilvio