robotology/mesh-support #1794: [Right thumb distal broken]

ddetommaso, 2024-04-23:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

During the robot calibration the right hand does not calibrate properly and it gets stuck.

### Detailed context

The issue appears without having noticed any critical problem in the previous sessions. It's important to mention that the problem seems recurring. 

### Additional context

[image]

https://github.com/robotology/icub-tech-support/assets/3440640/dd0929d0-0061-4845-b69c-0bcd5f625cf1

### How does it affect you?

The impact is severe because we need to start testing soon.

ddetommaso, 2024-04-30:
Please let us know a possible date of the intervention, possibly no later than the 10th of May.
cc @lornat75

AntonioConsilvio, 2024-05-13:
Hi @ddetommaso, @AntonioAzocar and I replaced the broken tendon in 9th May. We tested the robot and it works properly now!

https://github.com/robotology/icub-tech-support/assets/114915464/5e1ff1af-60ae-47ce-a164-eb33d3f65912

Closing!

closed 2024-05-13 by AntonioConsilvio