robotology/mesh-support #1795: R1 skirt is stuck

mga-97, 2024-04-23:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

R1 skirt, which is attached to the robot torso through a set of three elastic bands, does not come up smoothly when the torso elongates. Sometimes the skirt gets stuck in the lower position and obstructs the lidar view.

### Detailed context

- the elastic bands need to be replaced
- (optional) the rails needs some lubricant

AntonioConsilvio, 2024-05-06:
cc @fbiggi

fbiggi, 2024-05-20:
the three elastics have been replaced
tested the movement with @elandini84 

[image]
[image]

AntonioConsilvio, 2024-05-20:
Since the robot test was successfully completed, I proceed to closing the issue.

Well done @fbiggi! 🚀

cc @mga-97 @elandini84

closed 2024-05-20 by AntonioConsilvio