robotology/mesh-support #1797: Realsense not reachable or disconnected

SimoneMic, 2024-04-29:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

The realsense on the robot head doesn't stream data, nor is able to be launched by the YRI.

### Detailed context

Here the log of the YRI:
```
ergocub@ergocub-head:/usr/local/src/robot/robotology-superbuild/src/robots-configuration/ergoCubSN002$ yarprobotinterface --config realsense2.xml 
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/src/robots-configuration/ergoCubSN002/realsense2.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/src/robots-configuration/ergoCubSN002/realsense2.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  0.000119448 s
[INFO] |yarp.os.Port|/ergocub/yarprobotinterface| Port /ergocub/yarprobotinterface active at tcp://10.0.2.23:10082/
[INFO] startup phase starting...
[INFO] Opening device realsenseD450 with parameters [("robotName" = "/ergocub"), ("SETTINGS" [group] = "(depthResolution (640 480)) (rgbResolution (640 480)) (framerate 30) (enableEmitter true) (alignmentFrame RGB)"), ("HW_DESCRIPTION" [group] = "(clipPlanes (0.2 10.0))"), ("QUANT_PARAM" [group] = "(depth_quant 6)"), ("rotateImage180" = "false")]
[DEBUG] |yarp.dev.PolyDriver|realsenseD450| Parameters are (HW_DESCRIPTION (clipPlanes (0.200000000000000011102230246251565404 10.0))) (QUANT_PARAM (depth_quant 6)) (SETTINGS (depthResolution (640 480)) (rgbResolution (640 480)) (framerate 30) (enableEmitter true) (alignmentFrame RGB)) (device realsense2) (id realsenseD450) (robotName "/ergocub") (rotateImage180 false)
[ERROR] |yarp.device.realsense2| No device connected, please connect a RealSense device
[ERROR] |yarp.device.realsense2| Failed to initialize the realsense device
[ERROR] |yarp.dev.PolyDriver|realsenseD450| Driver <realsense2> was found but could not open
[WARNING] Cannot open device realsenseD450
[WARNING] Cannot open device realsenseD450
[INFO] Opening device RGBDWrapperyarp with parameters [("robotName" = "/ergocub"), ("period" = "0.03"), ("name" = "/depthCamera")]
[DEBUG] |yarp.dev.PolyDriver|RGBDWrapperyarp| Parameters are (device rgbdSensor_nws_yarp) (id RGBDWrapperyarp) (name "/depthCamera") (period 0.0299999999999999988897769753748434596) (robotName "/ergocub")
[INFO] |yarp.os.Port|/depthCamera/rpc:i| Port /depthCamera/rpc:i active at tcp://10.0.2.23:10083/
[INFO] |yarp.os.Port|/depthCamera/rgbImage:o| Port /depthCamera/rgbImage:o active at tcp://10.0.2.23:10084/
[INFO] |yarp.os.Port|/depthCamera/depthImage:o| Port /depthCamera/depthImage:o active at tcp://10.0.2.23:10085/
[INFO] |yarp.dev.PolyDriver|RGBDWrapperyarp| Created wrapper <rgbdSensor_nws_yarp>. See C++ class RgbdSensor_nws_yarp for documentation.
[WARNING] There was some problem opening one or more devices. Please check the log and your configuration
[ERROR] One or more devices failed opening... see previous log messages for more info
[INFO] Closing device RGBDWrapperyarp
[INFO] Closing device realsenseD450
[ERROR] Error in startup phase... see previous messages for more info
[WARNING] Interrupt # 1 # received.
[INFO] Interrupt received. Stopping all running threads.
[INFO] interrupt1 phase starting...
[INFO] interrupt1 phase finished.
[INFO] shutdown phase starting...
[INFO] Entering action level 5 of phase shutdown
[INFO] Executing detach action, level 5 on device RGBDWrapperyarp with parameters []
[ERROR] RGBDWrapperyarp is neither a wrapper nor a multiplewrapper, therefore it cannot have detach actions
[ERROR] Cannot run detach action on device RGBDWrapperyarp
[INFO] All actions for action level 5 of shutdown phase started. Waiting for unfinished actions.
[INFO] All actions for action level 5 of shutdown phase finished.
[WARNING] There was some problem running actions for shutdown phase . Please check the log and your configuration
[INFO] Closing device RGBDWrapperyarp
[INFO] Closing device realsenseD450
[INFO] shutdown phase finished.
[ERROR] Error in shutdown phase... see previous messages for more info
[INFO] |yarp.os.RFModule| RFModule failed to open.
```

carloscp3009, 2024-04-30:
Im going through some problems with the realsense too. In my case it starts working properly and a few second later it crashed with the following message:
```
...
[DEBUG] yarprobotinterface running happily
[ERROR] |yarp.device.realsense2| m_pipeline.wait_for_frames() failed with error: (Frame didn't arrive within 15000)
[ERROR] |yarp.devices.RgbdSensor_nws_yarp| Image not captured.. check hardware configuration
...
```

SimoneMic, 2024-05-09:
After updating the firmware of the realsense, this issue seems to have been solved.
After a few restarts and after a whole day, no issues were reported.

To update the firmware I manually downloaded the latest pack (5.16): https://dev.intelrealsense.com/docs/firmware-releases
And I've followed this guide from intel: https://dev.intelrealsense.com/docs/firmware-update-tool
Basically I've run this command (with the fw image file in the same directory where the command is ran): `rs-fw-update -f Signed_Image_UVC_5_16_0_1.bin`

I'll close the issue for the moment

closed 2024-05-09 by SimoneMic