robotology/mesh-support #1800: Wireless fault not working properly

elandini84, 2024-04-30:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

The wireless external fault button does not work. If the receiver board is connected, the robot is always in a fault state

### Detailed context

The robot cannot work with the external fault button connected to the robot. Yarprobotinterface log shows for all the joints the "External fault button pressed" error message even if the wireless button is not pressed.

### How does it affect you?

This situation does not allow us to use the external fault which is a valuable tool for our tests in Turin next week.

valegagge, 2024-05-06:
Hi @AntonioConsilvio,
 me and @andresoll we are aligned on this issue. Please contact us when you start working on it.

cc @Fabrizio69

AntonioAzocar, 2024-05-21:
Hi!
This morning, I and @andresoll  did the work on the robot and on disassembling the wiring we found this connector slightly detached from the fault board and that is why the fault was giving problems.
In fact, once it was reconnected properly, it no longer gave any problems and to prevent it from coming loose again I put a drop of glue on it.
Maybe Andrea will give more information

[image]

cc @Fabrizio69 @valegagge

andresoll, 2024-05-22:
thank you @AntonioAzocar, I asked @ste93  to prove that everything is working properly. This morning I will go to the robot arena to verify that everything is working. 

cc @valegagge

AntonioConsilvio, 2024-05-23:
Hi guys! Any feedback on this issue?

cc @AntonioAzocar @andresoll @ste93

closed 2024-05-23 by AntonioConsilvio