robotology/mesh-support #1801: Chest IMU cable connector broke

gabrielenava, 2024-05-03:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The chest IMU cable got pulled right before the connector, and the IMU does not start anymore.

### Detailed context

We tried with the external cable and the IMU starts. So I think it is something similar to what happened in https://github.com/robotology/icub-tech-support/issues/1689.

cc @DanielePucci 

### How does it affect you?

It blocks iRonCub experiments.

Gandoo, 2024-05-20:
[image]

[image]

Gandoo, 2024-05-20:
[image]

AntonioConsilvio, 2024-05-20:
Hi @gabrielenava! @Gandoo fixed the broken cable. The IMU has been tested and works properly now! 

> [!Note]
> We found the cables coiled inside the connector, they may have been coiled because it may look like the connector has to be inserted inside the IMU with a rotation. Actually, it should only be inserted (pushed in), so be careful not to rotate the connector as this could cause the cables to break inside!

Closing!

closed 2024-05-20 by AntonioConsilvio

gabrielenava, 2024-05-21:
thank you, I'll notify everyone not to rotate the connector, and hopefully this problem should not happen again!