robotology/mesh-support #1803: Torso Pitch goes in over temperature after few seconds

carloscp3009, 2024-05-08:
### Robot Name 🤖

ergoCub 1.1 S/N:002

### Request/Failure description

We moved the robot on the ground and we started the walking controller. The torso pitch immediately went in overheating 

https://github.com/robotology/icub-tech-support/assets/44415073/a3e34b1e-2484-4b13-9028-5743e220b2f2

### Detailed context

We check the current generated by the torso pitch but the board seems streaming 0A (the same for the PWM)

[image]

This is not happening for the other joints 

[image]

### Additional context

Here the mat file containing the dataset: 
[robot_logger_device_2024_05_08_11_16_53.zip](https://github.com/robotology/icub-tech-support/files/15246759/robot_logger_device_2024_05_08_11_16_53.zip)

And here the yarprobtinterface: 
[interface_overheating.txt](https://github.com/robotology/icub-tech-support/files/15246788/interface_overheating.txt)

### How does it affect you?

We have some demos the next week and a video shooting, without crane, tomorrow

cc @GiulioRomualdi @DanielePucci @S-Dafarra

GiulioRomualdi, 2024-05-10:
We have continuous overeating on the robot torso pitch. One happened during the demo and the robot felt.

This is the logger:  [overheating_torso2.txt](https://github.com/robotology/icub-tech-support/files/15272863/overheating_torso2.txt)

S-Dafarra, 2024-07-03:
It happened again twice today in front of visitors (https://github.com/ami-iit/lab-events-demos/issues/323). The torso pitch went to HF due to overheating twice, although very cols. In both cases, it was enough to idle/run the joint.

cc @maggia80 @MSECode @valegagge

MSECode, 2024-07-12:
I've checked the logs. The problem was due to the fact that the TDB stopped reading for more than 10 seconds (raw value -886) as shown by this log:
`[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=443s 6m 200u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=torso_pitch (NIB=1), Raw_temperature_value=-886)`

Then, the joint remained in fault, since no idle state got requested, but this is normal use case. However, checking the logs it is possible to observe that the tdb was correctly reading normal temperature values as shown here:
`[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=3877s 496m 216u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=torso_pitch (NIB=1), Raw_temperature_value=422)`

We are gonna investigate this behavior of failing to get the readings from the sensors at a certain point of time.

valegagge, 2024-07-12:
[Here](https://icub-tech-iit.github.io/documentation/temperature_sensors/software/dataflow/#error-handling) the documentation that explains the raw values used as error code.

closed 2025-12-24 by pattacini