robotology/mesh-support #1805: Issue with pinky of the right arm

ddetommaso, 2024-05-08:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During calibration the right hand take a while to start closing fingers and then, during the closing, the fingers get stuck.

### Detailed context

https://github.com/robotology/icub-tech-support/assets/3440640/0d3612b2-fb87-4d50-ae60-b271a80c0780

[log_icub-head_yarprobotinterface_2993.zip](https://github.com/robotology/icub-tech-support/files/15247737/log_icub-head_yarprobotinterface_2993.zip)

### How does it affect you?

We have an experimental study that is supposed to start beginning of the next week. Having the robot in this condition will cause a delay with data collection. Please we would ask you to schedule the intervention no later than the end of the next week.

maggia80, 2024-05-10:
@ddetommaso we are planning to do the intervention on May 14 morning. 

cc: @AntonioConsilvio

maggia80, 2024-05-15:
yesterday @fbiggi and @AntonioConsilvio fixed the issue.

ddetommaso, 2024-06-18:
The issue above recurred and the robot needs to be fixed possibly within the end of the month.

AntonioConsilvio, 2024-06-18:
Hi @ddetommaso, we will try to perform the maintenance service by the end of the month!

However, even though the problem is the same as this ticket (broken pinky tendon of the right arm), I would ask if you could kindly open a new issue.

Dividing maintenance work into different tickets allows us to keep track of problems on different robots (especially in the case of recurring problems like this one) and to study possible solutions so that they do not recur in the future.

Thank you in advance!

AntonioConsilvio, 2024-06-21:
Hi @ddetommaso, thank you for opening a new issue!

I am closing this one in favour of:

- https://github.com/robotology/icub-tech-support/issues/1836

closed 2024-06-21 by AntonioConsilvio