robotology/mesh-support #1807: Broken wire on the jetpack electronics

gabrielenava, 2024-05-09:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

A cable in the jetpack electronics is broken. It is the cable that connects the turbine to the fault button, so the turbine does not start.

I brought the electronics downstairs, would it be possible to fix it tomorrow morning?

cc @DanielePucci 

### Detailed context

It is the cable in the picture:

[image]

### How does it affect you?

It blocks iRonCub experiments

Fabrizio69, 2024-05-10:
Hi @gabrielenava , I'm looking for electronics to understand the problem. Where can I find it?

Gandoo, 2024-05-10:
ciao @gabrielenava il filo di Fault che era staccato è stato ripristinato.

[image]

[image]

cc @Fabrizio69 @AntonioConsilvio @DanielePucci

gabrielenava, 2024-05-10:
thank you @Gandoo!

we tested with a spool test and the turbine pump started this time, so the electronics should be fixed! we'll further verify this afternoon by turning on the turbines, but to me we can already close the issue!

closed 2024-05-10 by Gandoo